RockSat-X/RSXVT2026 #57: How should we do RF transmission?

PhucXDoan, 2025-08-29:
We need wireless communication between main and the vehicle; the vehicle will send sensor and image data back to the main electronics box so that we can (hopefully) recover it upon deintegration.

As of writing, my idea is to do the dumbest, simplest thing possible and just use an [ESP8266](https://www.adafruit.com/product/2471?srsltid=AfmBOorlEbe0qG78-xDZMUeX6KrnszpcuwmtBc7ogUfHz2mMv7yZnOke) so we can transmit over WiFi.

This approach has several advantages to be aware of:

1. _**It's easy to program.**_ As much as I don't like Arduino (or even the ESP for that matter), I doubt we have the capability to hand-roll something as complicated as the WiFi stack -- both hardware-wise and software-wise -- all in a single academic year.

2. _**It's easy to integrate.**_ The ESP8266 is just an all-in-one thing, so it should just be plug-and-play, and so we should be spending less time getting it to work. Furthermore, this also means it's easy to replace if anything were to happen.

3. _**Less crazy licensing.**_ With the ESP8266, it's just a WiFi device that's compliant with the regulation of FCC; the chances that it can be problematic will be slim. This doesn't mean we can skip the paperwork, however, but it will reduce the headache.

4. _**It's easy to do GPS rollout and RFI testing.**_ This I can not understate enough. If we have a more sophisticated RF system, then it needs to be designed to be verifiably working somehow. One of the tests that NASA does is GPS rollout and RFI testing where each university's experiment on the CarRoLL has their RF equipment (if any) enabled one at a time. This is so NASA can verify that the rocket's RF systems are not interfering with or by any experiment onboard.

Now what about the disadvantages?

1. _**We may need more transmission power.**_ We should design around a link of at least 100 meters, but whether or not the ESP8266 can handle this is not yet tested. Even if it was capable, this doesn't mean it'll perform the same in space with the other universities' experiments.

2. _**We may need higher throughput and reliability.**_ We need to transmit both sensor and image data, and the poor little ESP8266 might be able to handle it, especially if we roll up our sleeves and do some crazy data-compression. Nonetheless, the exact transmission speed we can get is an unknown, and there's a small chance we're going have to scope-up or downgrade data quality.

3. _**It's lame**_. This one is just my personal opinion honestly. It'd be cool if we can have a more advanced RF system, but that'll take some smart minds to tinker at. Besides, anything more than the ESP8266 will mean more specialized components (costs go up) and specialized knowledge in how to work with the system (effort goes up for the other non-RF people).

Let's see what we can come up with here.

shelbyoVT, 2025-10-07:
I know I should have probably figured this out mid last week but I did the ESP32 installation for Arduino. The error I received was that the port was busy or doesn't exist. My cable is capable of data transfer. AI told me that my problem was that I needed a driver, reading the ESP32 it would be the CP2102. Is that accurate, and where should I download this driver from?

PhucXDoan, 2025-10-07:
> The error I received was that the port was busy or doesn't exist.

IIRC, I gave you these [ESP-WROOM-32 dev-modules](https://www.amazon.com/ESP-WROOM-32-Development-Microcontroller-Integrated-Compatible/dp/B08D5ZD528) and the [XIAO ESP32-S3](https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html?srsltid=AfmBOoqUhI6A9wto-S4P6u9RzTFuGwHRc_GFkxtmIthmWeM6-ihSkgez). Notice how both have a **Reset** button and a **Boot** button.

<p align="center"><kbd>[image]</kbd></p>&nbsp;

I think what you have to do is press the Boot button to put the ESP into a mode where it can be programmed (this is what the bootloader does). The Arduino IDE should allow you to flash from there, and after the upload, press the reset button.

I don't think the CP2102 driver is necessary, because I didn't have to set that up, but then again, you could have a very different Windows installation from mine.

Don't stress too much about this though; I told you flashing the ESPs would be the hardest part :p.