RockSat-X/RSXVT2026 #111: Evaluating capabilities and limitations of OpenMV H7.

PhucXDoan, 2025-09-22:
We need to evaluate and document the capabilities and limitations of the [OpenMV H7 camera](https://www.digikey.com/en/products/detail/seeed-technology-co-ltd/102991322/11506484).

This mostly means figuring out the different CV algorithms that the OpenMV library provide as examples already and determining how accurate they are.

For instance, one of the algorithms that can be run on the OpenMV H7 is circle-fitting where circular objects in the image can be located at a (X, Y) coordinate. This circle-fitting algorithm can be a little janky and imprecise, however, so we'll need to look into what exactly causes inaccuracies and how we can improve upon it.

There might be other CV algorithms besides circle-fitting that we can potentially use for tracking; if they seem promising, we can explore that too.

PhucXDoan, 2025-10-01:
I'm going to give a more detailed list of "deliverables" here.

For Manan, since he's the one that has the camera right now, try looking into these things:

- Determine all the supported image formats that the OpenMV H7 camera can export. This means JPEG vs. raw, resolution size, color space, etc.
- For each of these image formats, determine their approximate size in bytes. There might be a way to just take a picture and then print out the picture's file size, but you can also probably automate this a bit by iterating through all the possible image formats, taking a picture, and then saving them to a uSD. Note that some image formats are fixed-size (e.g raw) but others are compressed (JPEG) and will vary; be scientific and document a bit on that if you can.
- With the different image formats, how do they affect the field-of-vision, if at all? Ideally, we want the smallest FOV, because this means we'll track the target from a farther distance; a huge FOV combined with low resolution might make things difficult.
- How does the camera handle bright light sources and dark lighting conditions? What if we there is a light source that's changing in brightness slowly vs. rapidly? How does the camera change its exposure? Can we control the camera's exposure level through code, or is it fixed by the hardware?

For people without the camera, meaning @ryan9182 and @justinvivadelli, here's some stuff we should look into:

- What are the available OpenMV cameras we could use? I know there's the OpenMV H7 and OpenMV H7 *Plus*, but to be honest, I have no idea what's the difference between them. I believe I gave Manan the original [OpenMV H7](https://www.digikey.com/en/products/detail/seeed-technology-co-ltd/102991322/11506484), but we need not use this exact model. If the board is red, it's probably compatible.
- Of the available cameras, which are practical to buy? The one I linked to DigiKey might be the only one we can actually by at a cheap price, because everything else is either out-of-stock or insanely marked up due to tariffs.
- What extension cables can we use? As of right now, the OpenMV H7 I gave to Manan has the camera mounted to the board, but if we want flexibility during integration, we should look into seeing if we can separate the camera from the board itself. I'm pretty sure the official OpenMV shop offers something, but I don't know how to use it.
- What different kinds of lenses can we use? As mentioned, we'd like a small FOV, so maybe a telescope lens is good to try out. I know there's also an IR lens, but I don't think it'll be that useful since there might be a lot of IR in space coming from the sun anyways (or maybe not? I'm spitballing there). We might end up just buying all the available lenses anyways, but it'll be nice to do some preliminary research here.
- Look into seeing if we can do some sort of simulation for the camera environment. For instance, I'm thinking it'll be really cool if we can create a Blender scene and be able to generate images that we can use to test the camera algorithms on (by loading on uSD card and having the OpenMV H7 parse and process them). Blender isn't that hard to learn the basics of, but I suspect doing good, realistic lighting to simulate the space conditions we might experience will be a bit of journey, but if someone is willing to put elbow grease into learning that, that'll be really cool.

Alright, that's everything on the top of my head right now. I'll comment more stuff if they come to me.

ryan9182, 2025-10-08:
Okay I researched all the Cameras and here's a summary of what I learned:

### In General

- All the cameras can have their lenses swapped out and all the original FOVS are pretty similar. There are two telescoping lens that reduces it from around 70 to 10 or 20
- Every camera is compatible with the extension cord on the OpenMV website
- I don’t think any other lenses are worth it. The IR and thermal lenses risk being overwhelmed by the sun even more than the regular lenses
- Global shutter is available for all modules and it might be something worth considering if blur is a problem.

### N6

- A lot of overkill for the simple object tracking we’re going to do. 
- Have to buy through kickstarter
- Does have the benefit of taking HDR pictures which could help with the sun
- Also has high resolution and we can decrease the fov which will allow us to track the object from very far away. 

### AE3

- Also pretty overkill on the compute
- Has dual CPUs which could be pretty cool and useful for separating tasks. For example we could potentially use it to do the calculations required to rotate the camera to follow the object
- Extremely low power consumption ~.30W
- Also have to buy through kickstarter

### RT 1062

- More compute heavy and power hungry than I think necessary 
- It also has worse frames at the same resolution. I’m not exactly sure why but I triple checked the specs from the website. I think it may have to do with the sensor used
- Has less resolution than the AE3 and N6 so tracking objects at a distance gets harder

### H7

- A little light on ram but shouldn’t be a problem for our purposes
- Has less resolution than the AE3 and N6 so tracking objects at a distance gets harder
- Cheaper
- High Framerate and decent compute 

### H7 Plus

- Better resolution but Draws more power
- Less frames than H7
- More compute than H7

Overall: 
I think that the AE3 is the best overall. It has better resolution, more memory, draws less power and is only $80 which is the cheapest out of all of the cameras besides H7. However, if we don’t want to buy more cameras I think the H7 works fine. The only concerning thing is the resolution at a distance and the lower ram (Which shouldn’t impact us too much). Otherwise all the cameras work with most of the attachments so upgrading is not too difficult. I think the N6 is obviously better overall but we’re paying $120 for better resolution and maybe some help with the sun in exchange for drawing 2 - 3x more power than the AE3. The only bad thing about the AE3 is the compute is overkill. There's 2 NPUs that we have almost no use for unless we use a neural network which is really unnecessary. Its also only available on kickstarter which is a little strange. 

Also heres a spreadsheet I made of some relevant data: 
[](https://docs.google.com/spreadsheets/d/1qODxmzkUFrhRwpfoHunBO3YmCYM7WDXOPijOCaC3o_A/edit?usp=sharing)